siderolabs/sbc-raspberrypi #8: Support option to disable console output for kernel

ruifung, 2024-05-04:
Having the kernel output to ttyAMA0 prevents usage of the UART interface.

Would be good to have an option to disable that.

ruifung, 2024-05-04:
Also to note: U-boot by default also outputs to serial console, I suspect this is what is causing issues when you have devices attached on the pi4's primary serial output.

ruifung, 2024-05-05:
[image]
I've been trying to patch u-boot to stop it writing output to serial, but all the instructions I find online can't seem to deal with this

frezbo, 2024-05-06:
I think its disabled using the bt overlay in `congix.txt` and passing `-console` args when generating the image

ruifung, 2024-05-06:
what about the u-boot output? I've managed to disable everything except the output from the initial u-boot loading, and I believe this overlay sets the bt overlay by default?

frezbo, 2024-05-06:
https://github.com/siderolabs/sbc-raspberrypi/blob/main/installers/rpi_generic/src/config.txt#L15 set this to `0`

ruifung, 2024-05-06:
I'll give that a try later then, but I was under the understanding that doing that would disable UART functionality from being accessible inside the OS...?

frezbo, 2024-05-06:
> I'll give that a try later then, but I was under the understanding that doing that would disable UART functionality from being accessible inside the OS...?

maybe try this https://github.com/raspberrypi/firmware/blob/master/boot/overlays/README#L3279, seems there's no overlay to completely disable serial, otherwise have to compile a custom u-boot

frezbo, 2024-05-06:
seems this might work: https://github.com/raspberrypi/firmware/blob/master/boot/overlays/README#L4781

ruifung, 2024-05-06:
I'm not trying to disable serial completely, just suppress all bootup output. But yeah. I guess I need to look into u-boot patches to try to suppress it's output.

closed 2025-11-19 by github-actions[bot]