siderolabs/sbc-raspberrypi #22: HDMI output shows wrong colors on v1.8

lwbt, 2024-10-02:
HDMI output shows wrong colors on v1.8. I compared it on several devices with v.1.7, where I can confirm that the colors are rendered properly.

[image]
[image]

samip5, 2025-02-09:
Are you referring to the Linux logo being blue as well?

lwbt, 2025-02-09:
> Hi [@lwbt](https://github.com/lwbt) Are you referring to the Linux logo being blue as well?

yes

samip5, 2025-02-09:
> > Hi [@lwbt](https://github.com/lwbt) Are you referring to the Linux logo being blue as well?
> 
> yes

It's not wrong per se, but not sure why it's blue. It changed with 6.12.x kernel I believe as it comes from the kernel.
The U-boot logo however is a different story.

closed 2025-11-19 by github-actions[bot]