siderolabs/sbc-raspberrypi #23: RPi5 is unable to boot

samip5, 2024-10-03:
Trying to boot Talos 1.8.0 with RPi5, results in just a mostly black screen with the u-boot logo on top right.

[image]

NOTE: I did try to use the UEFI loader for Pi5, and that is able to boot until it appears to lack network driver in kernel so it cannot properly boot even if we get into Talos.

smira, 2024-10-03:
There's no support for RPi5, and RPi5 is only supported with more recent Linux than the LTS 6.6. we're currently using

samip5, 2024-10-04:
> RPi5 is only supported with more recent Linux than the LTS 6.6

Not entirely correct, as mainline kernel lacks ethernet support on RPi5 due to the way it's handled there (RP1 chip is responsible for it, https://www.raspberrypi.com/news/rp1-the-silicon-controlling-raspberry-pi-5-i-o-designed-here-at-raspberry-pi/), so it can bare minimally boot on newer kernel, but that's pretty much it at the moment.

mooneydude, 2024-11-05:
Hi @samip5 any progress with maybe a uboot patch? It seems you are correct and
`head Makefile -n 4` on my RP5 running OS LTE 64 returns 

>VERSION = 6
>PATCHLEVEL = 6
>SUBLEVEL = 51

so my thinking is how do I stop RP1 from handling ethernet via patch?

samip5, 2024-11-05:
>Hi @samip5 any progress with maybe a uboot patch?

It would be best served to put uboot into the trash bin, and use systemd-boot instead for it.

> and i see talos 1.8.2 is 6.6.58 so my thinking is how do I stop RP1 from handling ethernet via patch?

You don't. The Pi5 architecture makes that impossible.

mooneydude, 2024-11-05:
> > Hi @samip5 any progress with maybe a uboot patch?
> 
> It would be best served to put uboot into the trash bin, and use systemd-boot instead for it.
> 
> > and i see talos 1.8.2 is 6.6.58 so my thinking is how do I stop RP1 from handling ethernet via patch?
> 
> You don't. The Pi5 architecture makes that impossible and Raspberry Pi OS LITE uses different kernel than Talos does.

I'm only using RPi OS Lite since i was able to install proxmox on it ([link](https://pimylifeup.com/raspberry-pi-proxmox/)) but I want talos on bare metal. Were you able to achieve this via systemd-boot?

samip5, 2024-11-05:
> I'm only using RPi OS Lite since i was able to install proxmox on it ([link](https://pimylifeup.com/raspberry-pi-proxmox/)) but I want talos on bare metal. Were you able to achieve this via [...]

Not yet, as haven't yet figured out the proper way to install UEFI loader first onto it. https://github.com/skyssolutions/talos-sbc-raspberrypi

RyandenOtter, 2025-01-09:
Any luck on this? I just got four Raspberry Pi 5s for Christmas, and I am dying to get a bare-metal Talos cluster running!

samip5, 2025-01-10:
> Any luck on this? I just got four Raspberry Pi 5s for Christmas, and I am dying to get a bare-metal Talos cluster running!

NVME boot doesn't work, SD card boot works and the mainline kernel talos uses cannot use ethernet.

brainplot, 2025-01-10:
Does Talos use only LTS kernel versions? Do we need to wait until the next LTS release is out for ethernet to work?

smira, 2025-01-10:
> Does Talos use only LTS kernel versions? Do we need to wait until the next LTS release is out for ethernet to work?

Yes, but at the moment the latest LTS 6.12.x Linux kernel is the latest Linux kernel release available as well, fyi.

samip5, 2025-01-10:
> Does Talos use only LTS kernel versions? Do we need to wait until the next LTS release is out for ethernet to work?

Will not work either. It's not upstreamed yet.

Sense545, 2025-01-11:
I got UEFI + NVME boot to work, but when kernel boots loop driver is missing.
Used worproject/rpi5-uefi and the dtb file from raspberrypi's repo (did not use u-boot).

Got this error while booting talos:
mrecovered from: early boot failed: error mounting "/dev/loop0": error mounting: 1 error(s) occurred: no such device

Has anyone managed to build talos using official/patched raspberrypi linux kernel?
Have not tried to set up kernel build with extra talos required config myself yet..

Cyclonit, 2025-01-12:
Hi,

I have three Raspberry Pi 5 to create a small bare metal k8s cluster. I'd love to use Talos from the start but I am struggling with understanding the current state of affairs. Obviously, there is no official support. Several people are trying to get Talos working on this hardware, with more or less success. I tried to follow what some others said, but couldn't make sense of it.

Given my setup as an example, could someone more knowledgeable explain if this is possible at all at the moment?

- 3x Raspberry Pi 5
- USB thumb drives as boot/os drive
- NVMEs for additional storage
- PoE for power
- connected via Ethernet

samip5, 2025-01-13:
> Hi,
>
> I have three Raspberry Pi 5 to create a small bare metal k8s cluster. I'd love to use Talos from the start but I am struggling with understanding the current state of affairs. Obviously, there [...]

For any of your peripherals to work:

- Requires the use of Raspberry Pi Foundation kernel
- nvme nor Ethernet are supported in mainline kernel 

they are provided via the RP11 IO microcontroller which is not supported fully at the moment.

nlindblo, 2025-01-14:
I also just bought 3x Raspberry Pi 5 with the GeeekPi M.2 NVMe, assuming I could join these to my existing Talos 1.9.0 cluster. I've spent the last day trying to figure out why I can't boot neither from sd card or the nvme and then I stumbled over this thread and realise it's not supported. I am now in 2 minds if I should make a return of the kit or wait, does anyone have an idea based on previous experience on how long to expect before the RB Pi 5 will get support, are we looking at weeks or months ?

Sense545, 2025-01-14:
I also plan on setting up 3 of them with NVMe.
I've put the project on hold for a bit, but I think the solution is to build a custom image with a custom Rasperry Pi kernel configured with some extra options that Talos needs.
There are some hints in this https://github.com/siderolabs/talos/issues/7978, but I have not tried building the kernel my self yet.

nlindblo, 2025-01-14:
I tried installing Ubuntu on one of them which booted fine on the nvme disk, perhaps running Talos in a kvm container could be a work around for the time being ?

Cyclonit, 2025-01-14:
I will go with a simple Ubuntu 24.10 running Kubernetes for now. Once Talos supports the Pi 5, I'll simply migrate the nodes one by one. This should not be a problem when using Kubernetes, because it does not care about the underlying OS.

samip5, 2025-01-14:
> are we looking at weeks or months ?

Months, possibly 6 months at least.

brainplot, 2025-01-15:
Looking at the past kernel LTS release schedule, I'd say we're looking at 9 to 11 months, assuming Talos strictly uses LTS releases.

| Version | Maintainer | Released | Projected EOL |
|---------|------------|----------|---------------|
| 6.12 | Greg Kroah-Hartman & Sasha Levin | 2024-11-17 | Dec, 2026 |
| 6.6 | Greg Kroah-Hartman & Sasha Levin | 2023-10-29 | Dec, 2026 |
| 6.1 | Greg Kroah-Hartman & Sasha Levin | 2022-12-11 | Dec, 2027 |
| 5.15 | Greg Kroah-Hartman & Sasha Levin | 2021-10-31 | Dec, 2026 |
| 5.10 | Greg Kroah-Hartman & Sasha Levin | 2020-12-13 | Dec, 2026 |
| 5.4 | Greg Kroah-Hartman & Sasha Levin | 2019-11-24 | Dec, 2025 |

Source: [Active kernel releases](https://www.kernel.org/category/releases.html)

samip5, 2025-01-15:
@brainplot  You're forgetting that full support has not landed in any version as of yet other than RPi Foundations own.

samip5, 2025-01-18:
Happy to report, we have a successful image.
Bult using Foundation kernel, and merging sideros kernel config on top.

Sources are in various repos under my Github org skyssolutions.

```
NODE   NAMESPACE   TYPE   ID        VERSION   SIZE     READ ONLY   TRANSPORT   ROTATIONAL   WWID                                   MODEL
       runtime     Disk   loop0     1         58 MB    true                                                                                        
       runtime     Disk   mmcblk0   1         64 GB    false       mmc                                                                             0xfb57fa56
       runtime     Disk   nvme0n1   1         256 GB   false       nvme                     eui.00000000000000007c35485240f9b1d4   TS256GMTE300S
```

samip5, 2025-01-18:
There's a discussion about the whole kernel thing: https://github.com/siderolabs/overlays/discussions/77

samip5, 2025-01-24:
You can get a working image for RPi5 by doing the following:
```
docker run --rm -t -v  $PWD/_out:/out -v /dev:/dev --privileged ghcr.io/samip5/siderolabs/imager:v1.9.2-skylab.30 rpi_5 --arch arm64 --overlay-image ghcr.io/skyssolutions/sbc-raspberrypi:v1.9.2-skylab.33 --overlay-name=rpi_5
```

philipptolksdorf, 2025-02-03:
@samip5 Thank you for your work!

**Current situation:**
Ethernet still isn't working on my machines. I managed to get a USB 3.0 adapter with a Realtek RTL8153 chip working, but my other adapter (ASIX AX88772) needs the usb-modem-drivers extension.

**What I've tried:**
I added the extension with --system-extension-image flag and while the build worked, the drivers won't load on a running system. 
Any ideas on how to get the extensions working?

**This happens when plugging in the ethernet usb adapter**
[image]

samip5, 2025-02-03:
> @samip5 Thank you for your work!
>
> **Current situation:**
> Ethernet still isn't working on my machines. I managed to get a USB 3.0 adapter with a Realtek RTL8153 chip working, but my other adapter [...]

System extensions are bound to specific kernel version (when they have kernel drivers) and I haven't built them as I simply skipped the extensions so that's why it doesn't work, if you look logs closely you can probably see module signature errors.

Ethernet does work with my image, but you need to go to UEFI settings and find Raspberry Pi settings there and change acpi to device tree.

samip5, 2025-02-04:
1.9.3: https://factory.kryptonian.dev/image/9952b075435d27130dca0fc30e0b9aad67ca5c87941d9a6803e36fe3b70870b3/v1.9.3-skylab.2/metal-arm64.raw.xz

nicolajv, 2025-02-06:
> 1.9.3 w/o extensions: https://factory.kryptonian.dev/image/c61ced8729442502e87d9e0ca1748b28f0269239bab2e38b46c70fcc10f7b8ef/v1.9.3-skylab.2/metal-arm64.raw.xz
> 
> NOTE: Requires going to UEFI settings and changing ACPI to device tree manually to get Ethernet.

Thanks for what you are doing here, this is awesome stuff 🙏 

I tried using this image by 1. booting into rpi-uefi from usb, 2. changing ACPI to device tree and 3. booting into the talos image on an SD card. However I'm also getting the error of time.cloudflare being unreachable, even when using a rtl8153 usb ethernet adapter 🤔 I'm wondering if I'm missing something in how this is supposed to go?

samip5, 2025-02-07:
> Thanks for what you are doing here, this is awesome stuff 🙏
>
> I tried using this image by
> 1. booting into rpi-uefi from usb,
> 2. changing ACPI to device tree and
> 3. booting into the talos image on an [...]

You're supposed to use the onboard Ethernet. I have not even tried any USB ones.

Peej11, 2025-02-08:
USB boot does seem to work with **very** basic testing

samip5, 2025-02-08:
> NVM - USB boot won't save your ACPI > Device Tree change

But it does if it's done before copying it over to USB as it seems to expect to be running on SD card.

samip5, 2025-02-13:
Updated to v1.9.4 for Pi5: <https://factory.kryptonian.dev/image/c61ced8729442502e87d9e0ca1748b28f0269239bab2e38b46c70fcc10f7b8ef/v1.9.4/metal-arm64.raw.xz>

NOTE: Requires going to UEFI settings and changing ACPI to device tree manually to get Ethernet.

regenscheid, 2025-03-06:
I'm happy to see the progress that's been made, but things look pretty bleak for the newer RPi 5s with the D0 stepping. It appears the rpi5-uefi project has shut down, at least until/unless the Foundation comes out with new firmware.

samip5, 2025-03-06:
It's quite sad about the D0 related problems with firmware.

Even the uefi project recommends to look elsewhere for an sbc. I'm glad I only have a single Pi5.

samip5, 2025-04-07:
This doesn't look too great for the future of PI5 with UEFI: https://github.com/raspberrypi/firmware/issues/1961

mohang6770, 2025-04-10:
Hi

Is anyone able to build talos on pi5, if yes could you please guide

Thanks in advance..

samip5, 2025-04-10:
> Hi
> 
> Is anyone able to build talos on pi5, if yes could you please guide
> 
> Thanks in advance..

factory.kryptonian.dev has builds for Pi5

nlindblo, 2025-04-10:
I been waiting for the pi5 support for some time now and the first time I heard it was available. If I create an image from factory.kryptonian.dev will it boot from nvme disk ? Is there any documentation on this ?

samip5, 2025-04-10:
> I been waiting for the pi5 support for some time now and the first time I heard it was available. If I create an image from factory.kryptonian.dev will it boot from nvme disk ? Is there any documentation on this ?

Currently it’s a bit problematic to install due to how the image needs to be flashed directly onto the nvme but then it should boot from it. I’m not sure what kind of docs you expect but there is no support for system extensions at the moment, simply because I haven’t built them. 

The images require going to UEFI settings and changing ACPI mode device tree to get Ethernet however.

I have been mentioning at least a few times that there’s a Talos version from my factory available but you might have missed it.

nlindblo, 2025-04-10:
Sorry, I’ve read comments but they have looked too complicated still and beyond my skills. So it sounds like when the system extensions becomes available (I need iscsi and Linux tools for Longhorn), I should be able to flash the nvme with the file from the factory, change from UEFI to ACPI and it should work. How would upgrades be managed in this scenario, would I need to “reflash” the nvme or can upgrades be done via talosctl still ?

samip5, 2025-04-10:
> Sorry, I’ve read comments but they have looked too complicated still and beyond my skills. So it sounds like when the system extensions becomes available (I need iscsi and Linux tools for Longhorn), [...]

Upgrades would work normal eg via talosctl. Just getting it on it is more involved.

nlindblo, 2025-04-10:
Excellent, I will have a go once the extensions are available. Thanks

mohang6770, 2025-04-11:
Hi

"Great work! I’ve created the Talos config and am attempting to deploy Cilium in proxyless mode. However, when I apply the configuration, the system restarts with errors. Has anyone else experienced this issue? 

```
`cluster:
  network:
    cni:
      name: none
  proxy:
    disabled: true`
```

Any help is appreciated

Thanks

samip5, 2025-04-11:
> Hi
> 
> "Great work! I’ve created the Talos config and am attempting to deploy Cilium in proxyless mode. However, when I apply the configuration, the system restarts with errors. Has anyone else experienced this issue?
> 
> ```
> `cluster:
>   network:
>     cni:
>       name: none
>   proxy:
>     disabled: true`
> ```
> 
> "Additionally, the interactive mode works for configuring it, but when I generate and apply the config file without making any changes, I encounter these errors."
> 
> ```
> `talosctl bootstrap --nodes 192.168.50.121
> error executing bootstrap: rpc error: code = Unavailable desc = connection error: desc = "transport: authentication handshake failed: tls: failed to verify certificate: x509: certificate signed by unknown authority (possibly because of \"x509: Ed25519 verification failure\" while trying to verify candidate authority certificate \"talos\")"
> 
> talosctl kubeconfig -n 192.168.50.121 -f
> error copying: rpc error: code = Unavailable desc = connection error: desc = "transport: authentication handshake failed: tls: failed to verify certificate: x509: certificate signed by unknown authority (possibly because of \"x509: Ed25519 verification failure\" while trying to verify candidate authority certificate \"talos\")"
> `
> ```
> 
> Any help is appreciated
> 
> Thanks

Those error suggest that you don’t have the right talos config for the cli with the right certficates so you’re unable to talk to it.

mohang6770, 2025-04-11:
A huge thank you to everyone! I've successfully got Cilium (CNI) working in proxyless mode on the Raspberry Pi 5 with the control plane and worker configuration. Once I’ve wrapped everything up, I’ll create a guide in case it’s useful to anyone. The only thing left to do is test the upgrade once the 1.9.5 release is available.

samip5, 2025-04-12:
> The only thing left to do is test the upgrade once the 1.9.5 release is available.

When preparing that, hit a little snag with pushing container images but should hopefully be available soon.

nlindblo, 2025-04-12:
Will that include extensions as well, or is that still TODO ?

samip5, 2025-04-12:
> Will that include extensions as well, or is that still TODO ?

Some extensions, for Longhorn support.

samip5, 2025-04-13:
Okay, so the newer version of the image has problems with kernel as rp1 is unable to initialize needed things like the PCIe bus.
This is like playing whack a mole which I'm not really fond of.

samip5, 2025-04-13:
I'm sorry to announce that I will be abandoning building images for Pi5 after hitting multiple snags with the kernel build and there's simply no good ways to automate this patching needed.
The maintenance burden is more than I would like it to be for a single person, and I'm spending actual money on the builds (not much, but still actual money due to building on AWS).

Factory will be taken offline on 30th of April.

The building of this basically requires:
- Custom Image Factory (https://github.com/siderolabs/image-factory/compare/main...skyssolutions:siderolabs-image-factory:skylab)
- Kernel (https://github.com/skyssolutions/talos-kernel)
- Forked Talos (https://github.com/skyssolutions/talos)
- sbc-overlay (https://github.com/skyssolutions/talos-sbc-raspberrypi, includes packaging rpi5-uefi)
- overlays repo (https://github.com/skyssolutions/siderolabs-overlays)
- boot-assets automation (https://github.com/samip5/talos-boot-assets, here you can find the full workflows on making images)

rothgar, 2025-04-13:
Thank you @samip5 for taking the time to maintain this and sharing the steps.

We'd encourage anyone interested in making this work in a sustainable way to contribute to [u-boot](https://www.u-boot.org/) as it will give Talos the widest possible support for SBCs now and in the future.

There is still not mainline support for the raspberry pi 5 in the Linux Kernel which is another blocker for us to make these devices fully functional.

mohang6770, 2025-04-15:
@samip5 

Please let us know if we can contribute to the expenses or help in any other way to ensure this isn’t abandoned.

samip5, 2025-04-15:
@mohang6770 Thank you for the offer. We can probably do an open-collective which would allow for transparency on the funds usage. It's mostly that I need help with keeping up with Talos & Foundation kernel changes if we want to maintain this. I only have a single RPi5 currently (which I also need for other things) and testing changes for it are somewhat non-trivial as we cannot test on non-physical hardware due to needing to test if kernel is able to properly initialize the devices like USB, PCie bus.

I think the effort should be mostly directed at getting u-boot support extended but as that's coded in C and very low-level, it's outside of my wheel house.

The problems I encountered were that the rp1 fails to initialize when using the defconfig for Pi5 on the latest release of Foundation kernel. I couldn't even get UART console to print the kernel messages to it, so that I could get proper logs on what's not working in kernel.

mohang6770, 2025-04-15:
@samip5 

I'll help you in any way I can, though I'm not a developer and don't have an understanding of low-level coding. I really appreciate the work you're doing, and I'm happy to support however possible—whether that's through testing, providing feedback, or helping with setting up something like Open Collective to make things a bit easier on your end. Let me know what would be most helpful!

lehmanju, 2025-04-15:
There are people already working on extending U-Boot support: https://lore.kernel.org/u-boot/cover.1738749764.git.oleksii_moisieiev@epam.com/

samip5, 2025-04-16:
It seems that Open Collective Europe requires at least two admins on the collective, so I would need someone for that (preferably from EU), and I have moved the repositories to a new org on Github: https://github.com/talos-on-rpi5.

I would also need at least one but preferably two who can help with merging the upstream changes (be it siderolabs change, or Raspberry Pi foundations) when those are needed as well as few who have an Pi5 and would be willing to test things (which could get some funds from the said collective for doing so that their time/help is not for free).

How does this sound?

samip5, 2025-04-16:
> A native way to supply a custom kernel via an overlay would definitely help, but I don't think it would completely solve the problems. Right now kernels and kernel modules are signed with the same key so they need to be built at the same time (e.g. system extensions).
> 
> We've talked about how we could do this in the past, but never came to a complete solution. It's not only a rpi request, there are other use cases (e.g. realtime kernel).
> 
> If you want to start a github discussion about custom kernels via overlays we can restart the discussions and think through the current limitations.

There already is a discussion about the RPi kernel in the overlays discussions. https://github.com/siderolabs/overlays/discussions/77

pndrik, 2025-04-28:
@samip5 I'm a bit limited on time currently, but I'd be happy to support you on your efforts. I’m based in Switzerland.

**My setup:** I’m running a custom build of Talos 1.9 on a few Raspberry Pi CM5s, using NVMe for storage. A couple of caveats:
- I had to bypass U-Boot and have the CM 5s loading the kernel directly.
- I can only get network working on the RPi kernel tag of "stable_20241008" [1] (Linux 6.6.51). On newer tags, the device is visible but can’t communicate.

**Future:**
- There is promising progress on the upstream Linux kernel [2].
- I suspect upstream will move faster, but if I can carve out some time, I'll dig into the networking situation on newer RPi kernel tags as well as the NVMe one with U-Boot.

[1] https://github.com/raspberrypi/linux/tree/stable_20241008
[2] https://lore.kernel.org/linux-pci/cover.1738963156.git.andrea.porta@suse.com/

mohang6770, 2025-04-29:
@samip5 
Please let me also know how to the setup and i  will also test

adrien-jeser-doctolib, 2025-04-29:
I have a raspberry pi 5 and I live in France. I work on low level (PCSC and GPU) and linux kernel.

If needed, I can help.

[80 more replies]