siderolabs/sbc-raspberrypi #30: HDMI output stuck at 640x480 due to `hdmi_safe=1'

brujoand, 2024-12-23:
Over [here](https://github.com/siderolabs/sbc-raspberrypi/blob/main/installers/rpi_generic/src/config.txt#L6 ) we are setting:
`
# Enable maximum compatibility on both HDMI ports;
# only the one closest to the power/USB-C port will work in practice.
hdmi_safe:0=1
hdmi_safe:1=1
`
Which according to [RPIconfig](https://elinux.org/RPiconfig) means:
> Use "safe mode" settings to try to boot with maximum hdmi compatibility. This is the same as the combination of: hdmi_force_hotplug=1, hdmi_ignore_edid=0xa5000080, config_hdmi_boost=4, hdmi_group=2, hdmi_mode=4, disable_overscan=0, overscan_left=24, overscan_right=24, overscan_top=24, overscan_bottom=24 

Specifically this the `hdmi_mode=4` causes the resolution to be fixed at `640x480`. Is this intentional if not could we change this to instead use the expanded options but drop the hdmi_ignore_edid, hdmi_group and hdmi_mode to let the monitor an the pi figure this out on their own?

frezbo, 2024-12-23:
PR's are welcome

brujoand, 2024-12-23:
So for my specific usecase I just added the overlay options to completely override the config.txt and instead of the `hdmi_safe:x=1` I only used `hdim_force_hotplug:x=1` instead. And kept the rest of the default config.txt

If we want these SBC's to work just like any other PC I'm thinking this is what should be the default, and I could add a PR to that effect. 

However, adding these options when calling out to the factory for the image is fairly trivial. The current defaults are at least very likely to always work, even if the resolution is limited. And there might be other constellations like compute boards etc that might behave differently . Thus, my vote is actually on keeping this defensive default as is. 

For any late adventurers  finding this in the future, what I did was just creating this overlay config:
```
overlay:
  name: rpi_generic
  image: siderolabs/sbc-raspberrypi
  options:
    configTxt: |
      gpu_mem=32
      hdmi_force_hotplug:0=1
      hdmi_force_hotplug:1=1
      kernel=u-boot.bin
      arm_64bit=1
      arm_boost=1
      enable_uart=1
      dtoverlay=disable-bt
      dtoverlay=disable-wifi
customization:
  systemExtensions:
    officialExtensions: []
```
Throwing it at the factory: `curl -X POST --data-binary @assets/rpi-overaly.yaml https://factory.talos.dev/schematics`
Which gave me an image id to use for the installer. And it just worked. :magic_wand:

closed 2024-12-24 by brujoand