siderolabs/sbc-raspberrypi #35: Installer doesn't respect Talos installDisk selector

samip5, 2025-01-19:
It always just installs talos to the boot drive, no matter what one tries.
Looking at the code, it's likely an issue with installer?

frezbo, 2025-01-19:
This is as expected, writing a raw disk means talos is already installed to that disk

closed 2025-01-19 by frezbo

samip5, 2025-01-19:
> This is as expected, writing a raw disk means talos is already installed to that disk

Let's say, you are trying to install onto a nvme, without access to an nvme dock, that's difficult without it respecting it?

frezbo, 2025-01-19:
> Is that how it works on eg, Ampere ARM64? If it doesn't like I suspect, why are pis different in that aspect?

One proper ARM servers you can PXE boot or boot from USB/Virtual media without writing talos to a disk

samip5, 2025-01-19:
> > Is that how it works on eg, Ampere ARM64? If it doesn't like I suspect, why are pis different in that aspect?
> 
> One proper ARM servers you can PXE boot or boot from USB/Virtual media without writing talos to a disk

Let's ignore both of those options aka PXE and USB/virtual media so you're saying that people run it on Ampere without writing it to disk? Even state?

Well, in any case, I think this is to the right way to manage to respect it? https://github.com/skyssolutions/talos-sbc-raspberrypi/blob/rpi5/installers/rpi_5/src/main.go#L51

frezbo, 2025-01-19:
> Let's ignore both of those options aka PXE and USB/virtual media so you're saying that people run it on Ampere without writing it to disk? Even state?

no, I meant that when booting off virtual media, talos doesn;t  have STATE partitons created already, but when writing a raw disk it's already there, so `.machine.install` is ignored, I agree it's not ideal (we would probably change that in the future)

> Well, in any case, I think this is to the right way to manage to respect it? https://github.com/skyssolutions/talos-sbc-raspberrypi/blob/rpi5/installers/rpi_5/src/main.go#L51

but this is only writing boot assets