siderolabs/sbc-raspberrypi #36: Pis are able to boot without u-boot/systemd-boot

samip5, 2025-01-22:
There should be a way to do tryboot things without u-boot or systemd-boot.
However that means that the kernel and initramfs need to be in the boot partition so Pis bootloader can load them and possibly compressed differently for initramfs.

Currently the kernel seems to be stored in the rootfs's root, if not mistaken.

Related: https://www.raspberrypi.com/documentation/computers/config_txt.html#example-update-flow-for-ab-booting

frezbo, 2025-01-22:
Talos will not support this since Talos always follows A/B install schema and it's not possible without a proper bootloader

samip5, 2025-01-23:
> Talos will not support this since Talos always follows A/B install schema and it's not possible without a proper bootloader

Pis do support A/B booting without U-boot or systems-boot as the bootloader itself in them support it.

frezbo, 2025-01-23:
> > Talos will not support this since Talos always follows A/B install schema and it's not possible without a proper bootloader
> 
> Why wouldn't this be possible? It requires two fat32 boot partitions to achieve and some config changes.
> 
> Pis do support A/B booting without U-boot or systemd-boot as the bootloader itself in them support it so there is no absolute need for 3rd party bootloader to be used.

from Talos perspective we need a proper bootloader to have custom cmdline options and boot menu, so we've have to wait until things are upstreamed

samip5, 2025-01-23:
> have custom cmdline options and boot menu

1. Pis don't need a boot menu IMO
2. Custom cmdline is for this reason usually baked into kernels default cmdline with other distributions.

frezbo, 2025-01-23:
this is all fine if Talos was just for RPI's , but Talos has to support other platforms as well

samip5, 2025-01-23:
> this is all fine if Talos was just for RPI's , but Talos has to support other platforms as well

Ah, right because the default cmdline in the kernel might not be desired on other platforms.

closed 2025-11-19 by github-actions[bot]