siderolabs/sbc-raspberrypi #38: Talos v1.9.3 fails to boot on Raspberry Pi 4B

IAMSolaara, 2025-02-03:
## Bug Report
Talos v1.9.3 fails to boot on Raspberry Pi 4B
### Description
I have tried upgrading one of my pi control plane nodes to talos v1.9.3 using the following command:
`talosctl upgrade --preserve -n cp03 -i factory.talos.dev/installer/69c0f32290e5f1abe266a219fbee33ff9f659fd49820452e47d626b86e292b4a:v1.9.3 --stage`.

It can successfully install the upgrade into the B partition and I look into the monitor for output but it ends up bootlooping after a hard reset past the "EFI stub: Decompressing Linux kernel...".

I tried doing the same but flashing a v1.9.3 image obtained via the Factory using that same schematic (the same one I have on v1.9.2) and got the same results.

Flashing the SD card back with v1.9.2 got the node working again as it should. I don't have any UART devices to do much further :/

I noticed that when normally booting the little tuxes on the top of the screen look blueish; reminds me of the Pi 5 16GB boot issues. Maybe that could have something to do here? even though this is a Pi 4B 4GB...

I have also tried upgrading to the latest EEPROM (which I had already done when I got the pi's) and no change there.

Unfortunately rebooting to partition A via GRUB doesn't help at all meaning there might be some issue with something like firmware in the overlay.

However flashing back to a fresh image for v1.9.2 does fix it.

### Logs
None
### Environment

- Talos version: [`talosctl version --nodes <problematic nodes>`] Can't really get this info.
- Kubernetes version: [`v1.32.0`]
- Platform: Raspberry Pi 4B

smira, 2025-02-03:
I wonder if that is https://github.com/siderolabs/sbc-raspberrypi/pull/34, so it can be tested by overriding the `config.txt` in the schematic to be default one.

IAMSolaara, 2025-02-03:
I'm not at home right now so it's a bit hard for me to check on the pi's, but I can confirm the fresh v1.9.2 image does **not** have that watchdog thing enabled:
```
$ k debug node/cp03 -it --image=fedora --profile=sysadmin -n kube-system
Creating debugging pod node-debugger-cp03-k5ndb with container debugger on node cp03.
If you don't see a command prompt, try pressing enter.
[root@cp03 /]# mount /dev/mmcblk0p1 mnt
[root@cp03 /]# cd mnt
[root@cp03 mnt]# cat config.txt
# See https://www.raspberrypi.com/documentation/computers/configuration.html
# Reduce GPU memory to give more to CPU.
gpu_mem=32
# Enable maximum compatibility on both HDMI ports;
# only the one closest to the power/USB-C port will work in practice.
hdmi_safe:0=1
hdmi_safe:1=1
# Load U-Boot.
kernel=u-boot.bin
# Forces the kernel loading system to assume a 64-bit kernel.
arm_64bit=1
# Run as fast as firmware / board allows.
arm_boost=1
# Enable the primary/console UART.
enable_uart=1
# Disable Bluetooth.
dtoverlay=disable-bt
# Disable Wireless Lan.
dtoverlay=disable-wifi

```

I tried mounting the v1.9.3 image and it does contain that line to enable the watchdog. I'll try making an image without it and get that pi upgraded.

IAMSolaara, 2025-02-03:
Ok, I got an image made with the old config and it boots!:

[image]

[image]

This is the schematic I used:
```yaml
overlay:
  image: siderolabs/sbc-raspberrypi
  name: rpi_generic
  options:
    configTxt: |
      # See https://www.raspberrypi.com/documentation/computers/configuration.html
      # Reduce GPU memory to give more to CPU.
      gpu_mem=32
      # Enable maximum compatibility on both HDMI ports;
      # only the one closest to the power/USB-C port will work in practice.
      hdmi_safe:0=1
      hdmi_safe:1=1
      # Load U-Boot.
      kernel=u-boot.bin
      # Forces the kernel loading system to assume a 64-bit kernel.
      arm_64bit=1
      # Run as fast as firmware / board allows.
      arm_boost=1
      # Enable the primary/console UART.
      enable_uart=1
      # Disable Bluetooth.
      dtoverlay=disable-bt
      # Disable Wireless Lan.
      dtoverlay=disable-wifi
customization:
  systemExtensions:
    officialExtensions:
      - siderolabs/util-linux-tools
```

nberlee, 2025-02-03:
Based on what @IAMSolaara is showing yes....

However, over the weekend I try to replicate it as I was made aware of this issue here https://github.com/siderolabs/talos/issues/9981#issuecomment-2626682219

But I cannot replicate it on a fresh image (without any extensions). Which bothers me a bit :)

smira, 2025-02-03:
Ok, #34 got reverted, next Talos release should pick it up.

closed 2025-02-04 by talos-bot

IAMSolaara, 2025-02-05:
Hello, I got myself a UART serial cable thingy to see if I could've dug deeper.
I tried making a new image with the old overlay _with_ the watchdog overlay enabled and even there doesn't get me any further:
```
  Booting `A - Talos v1.9.3'

EFI stub: Decompressing Linux Kernel...

U-Boot 2024.07 (Mar 01 2024 - 17:26:00 +0000)

DRAM:  992 MiB (effective 3.9 GiB)
RPI 4 Model B (0xc03115)
Core:  210 devices, 16 uclasses, devicetree: board
MMC:   mmc@7e340000: 0
Loading Environment from FAT... Unable to read "uboot.env" from mmc0:1...
In:    serial,usbkbd
Out:   serial,vidconsole
Err:   serial,vidconsole
Net:   eth0: ethernet@7d580000

PCIe BRCM: link up, 5.0 Gbps x1 (SSC)
starting USB...
Bus xhci_pci: Register 5000420 NbrPorts 5
Starting the controller
USB XHCI 1.00
scanning bus xhci_pci for devices... 2 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found
Hit any key to stop autoboot:  0
U-Boot>
U-Boot>
```

nberlee, 2025-02-06:
After seeing this report https://github.com/siderolabs/talos/issues/9981#issuecomment-2637957324

Are you using a SDcard @IAMSolaara , then I know how to replicate it, and looking to fixing the underlying problem. As I solely tested on Sata USB (running 24/7) and thumb-drives to try to replicate this problem...

IAMSolaara, 2025-02-06:
Yes, I am using an SD card.

PR #40 (2025-02-04): release(v0.1.3): prepare release
This is the official v0.1.3 release.

Fixes #38