siderolabs/sbc-raspberrypi #43: Talos doesn't boot from SD Card on Raspberry PI 4B 8GB

panoukos41, 2025-02-08:
# Bug
I have been trying to talos to boot from my SD Card. The usb boot works fine but I can't add a second drive for data operations as is recommended since the PI most of the times selects the wrong drive.

# Versions tried
- 1.8.4
- 1.9.0 to 1.9.3

# Steps tried
1. Tried getting a bigger SD Card (128GB) as I thought the problem was the size.
2. I tried solutions decribed here #38 but still the same.

# Setup
- On windows using the `Raspbery PI Imager` software for both SSD and SD Card.
- Latest bootloader with SD Card configured as the first source.

Image of RPI trying to boot:

[image]

The Raspberry PI OS boots correctly from the SD Card.

samip5, 2025-02-09:
Please show us your boot partitions contents? Should be as simple as mounting the SD card on Windows as it's a FAT32 partition.

samip5, 2025-02-09:
Also Talos is less than 1GB in total size for the kernel + initramfs + boot blobs.

panoukos41, 2025-02-09:
Windows Disk Management and DiskPart won't let mount to see the contents I just get errors, know any other softwrare? Or any linux commands to see it on the PI? I used the ssd to have a light OS so that I could use the dd tool there as in the guide which still resulted in the same behavior.

panoukos41, 2025-02-09:
Trying to mount the the `sdb2, sdb3, sdb4` resulted in this error

[image]

while the `sdb1` has these as contents

[image]

samip5, 2025-02-10:
> Trying to mount the the `sdb2, sdb3, sdb4` resulted in this error
> 
> [image]
> 
> while the `sdb1` has these as contents
> 
> [image]
> 
The other partitions are xfs, so the host you're using to mount them might lack the filesystem tools for supporting that. 

Please see the contents of config.txt there as you should have gotten as far as u-boot.

samip5, 2025-02-10:
Okay, so that is indeed right. 
They were saying in Slack that there were some corruption in their cache for Image Factory so that might have broken it? @smira could this be the case of cache corruption?

smira, 2025-02-10:
I have no idea what you're referring to.

samip5, 2025-02-10:
> I have no idea what you're referring to.

https://taloscommunity.slack.com/archives/CMARMBC4E/p1738749446732549?thread_ts=1738738076.919479&cid=CMARMBC4E

"OK, it looks as if the local caching registry we have for image-factory had cached it corrupted somehow. We've cleared it out and now it builds & pulls cleanly for me." - Tim Jones, 5th of Feb

roukydesbois, 2025-02-18:
Hello!

I'm trying to do the same (boot Talos 1.9.3) on my RPI4 8Gb, and can't have it boot as the author of this issue.

I will add that I have the "Kernel not found" 7 LED flashes

frezbo, 2025-02-19:
1.9.4 should work, previous 1.9.x versions are known to have issues booting

roukydesbois, 2025-02-19:
I just used with 1.9.4 (using [https://factory.talos.dev/image/d4840a63be0bbe7c7f3f6c120effec6f6c5e28831a91936a303904cab4f80958/v1.9.4/metal-arm64.raw.xz - ](https://factory.talos.dev/?arch=arm64&board=rpi_generic&cmdline-set=true&extensions=-&extensions=siderolabs%2Fbtrfs&extensions=siderolabs%2Fiscsi-tools&extensions=siderolabs%2Ftailscale&extensions=siderolabs%2Futil-linux-tools&platform=metal&target=sbc&version=1.9.4) and still have the same issue!

panoukos41, 2025-02-19:
I think it might be related to uboot in the end as a quick search resulted in many forums talking about problems with the RPI 8GB.

samip5, 2025-02-19:
> I just used with 1.9.4 (using [this image configuration](https://factory.talos.dev/?arch=arm64&board=rpi_generic&cmdline-set=true&extensions=-&extensions=siderolabs%2Fbtrfs&extensions=siderolabs%2Fiscsi-tools&extensions=siderolabs%2Ftailscale&extensions=siderolabs%2Futil-linux-tools&platform=metal&target=sbc&version=1.9.4) and still have the same issue!

Just tested with that image on a SD card and it does boot fine? Which bootloader version are people running that have this issue? Mine is working with version 8ba17717 / 2023/01/11

roukydesbois, 2025-02-19:
Alright - after @samip5 's comment I flashed a raspbian, went into raspi-config and set the bootloader version to the default one (not the latest), and it booted! Thanks @samip5 !

panoukos41, 2025-02-19:
Okay can confirm it works just fine by switching to the default bootloader.

Thanks @roukydesbois and @samip5 😄

closed 2025-02-19 by panoukos41

jhart99, 2025-02-23:
To further round out this issue.  It seems that the issue is in the upstream bootloader code.  Linking the[ issue ](https://github.com/raspberrypi/rpi-eeprom/issues/668) here in case anyone else experiences this issue in the future.

QuinnHerden, 2025-03-12:
I ran into the same issue on my Raspberry Pi 4B 4GB.
I also had success with altering the boot loader to an earlier version.

DijegoDev, 2025-03-19:
I got 3x Raspberry Pi 4 Model B - 8GB
bootloader: 8ba17717 2023/01/11
update-ts: 1674059736

I downloaded Talos 1.9.x versions. Put all of them at some point on the SD cards and on a Flash Drive. But none of them boot. When I connect the raspberry pi to a screen it shows the same messages as seen on the original post.

At some point I loaded raspbian, changed the boot to default and set it to pick from USB, but it still won't work. What am I doing wrong? When I tried Talos for the first time a while back I was able to just put an image on the SD card and it worked.

samip5, 2025-03-19:
> I got 3x Raspberry Pi 4 Model B - 8GB bootloader: 8ba17717 2023/01/11 update-ts: 1674059736
>
> I downloaded Talos 1.9.x versions. Put all of them at some point on the SD cards and on a Flash Drive. [...]

Did you try 1.9.5?

DijegoDev, 2025-03-20:
> Did you try 1.9.5?

Yes, I tried 1.9.0, 1.9.1, 1.9.2, 1.9.3, 1.9.4, 1.9.5 from the factory. No extensions.

AlessandroZanatta, 2025-03-26:
I'm on a PI4 with 4GB, having the same issues (green led blinking 7 times - issues with the kernel image) on Talos 1.9.4 using the base configuration on the [official documentation](https://www.talos.dev/v1.9/talos-guides/install/single-board-computers/rpi_generic/#download-the-image). 

I've tried @roukydesbois solution, but I'm afraid it no longer is viable. Currently, it appears that both `default` and `latest` point to the same bootloader version (75c1e570e99e14684e79e6ab8f5355d831440c71, 2025/02/11), which may very well be the culprit. 

```
$ sudo rpi-eeprom-update
BOOTLOADER: up to date
   CURRENT: Tue 11 Feb 17:00:13 UTC 2025 (1739293213)
    LATEST: Tue 11 Feb 17:00:13 UTC 2025 (1739293213)
   RELEASE: default (/usr/lib/firmware/raspberrypi/bootloader-2711/default)
            Use raspi-config to change the release.

  VL805_FW: Dedicated VL805 EEPROM
     VL805: up to date
   CURRENT: 000138c0
    LATEST: 000138c0
```

Could someone who fixed this issue using @roukydesbois solution please share the bootloader version they are on? Moreover, does anyone know of a (safe) way to downgrade the bootloader to a specific version? `raspi-config` does not appear to support it, nor I can find it documented anywhere, and I do not know enough about this.

samip5, 2025-03-26:
> I'm on a PI4 with 4GB, having the same issues (green led blinking 7 times - issues with the kernel image) on Talos 1.9.4 using the base configuration on the [official documentation](https://www.talos.dev/v1.9/talos-guides/install/single-board-computers/rpi_generic/#download-the-image). 
> 
> I've tried @roukydesbois solution, but I'm afraid it is no longer viable. Currently, it appears that both `default` and `latest` point to the same bootloader version (75c1e570e99e14684e79e6ab8f5355d831440c71, 2025/02/11), which may very well be the culprit. 
> 
> ```
> $ sudo rpi-eeprom-update
> BOOTLOADER: up to date
>    CURRENT: Tue 11 Feb 17:00:13 UTC 2025 (1739293213)
>     LATEST: Tue 11 Feb 17:00:13 UTC 2025 (1739293213)
>    RELEASE: default (/usr/lib/firmware/raspberrypi/bootloader-2711/default)
>             Use raspi-config to change the release.
> 
>   VL805_FW: Dedicated VL805 EEPROM
>      VL805: up to date
>    CURRENT: 000138c0
>     LATEST: 000138c0
> ```
> 
> Could someone who fixed this issue using @roukydesbois solution please share the bootloader version they are on? Moreover, does anyone know of a (safe) way to downgrade the bootloader to a specific version? `raspi-config` does not appear to support it, nor I can find it documented anywhere, and I do not know enough about this.

Just to test, can you try https://factory.kryptonian.dev/image/c61ced8729442502e87d9e0ca1748b28f0269239bab2e38b46c70fcc10f7b8ef/v1.9.4/metal-arm64.raw.xz ?

That's using foundation's kernel and main target is Pi5 but that schematic is the generic overlay.

samip5, 2025-03-26:
> an untrusted source

I get it, but the source for that is freely available on Github, under the this org: https://github.com/skyssolutions.

Thanks for testing anyways, so it is 100% the bootloader. This problem has been raised in the firmware repo: https://github.com/raspberrypi/firmware/issues/1953

AlessandroZanatta, 2025-03-27:
I've managed to fix the issue. I'll report the steps below, they may help someone (@DijegoDev may be interested).

First, I've updated everything through raspian. First, perform a full upgrade `sudo apt update && sudo apt full-upgrade`. Then, update the PI itself: `sudo rpi-update`. Bootloader version did not change, but I believe the firmware got updated (whatever that truly means, pardon my ignorance). I'm not sure whether these steps are really required, but I'm documenting them here in case they are needed. 

Then, I've written to SD the (base) Talos image. I've tried it on the PI, but it still did not work. Hence, I've tried to copy to the boot partition (`/dev/sda1` for me) the [new firmware](https://github.com/raspberrypi/firmware/releases/tag/1.20250326) (thanks @samip5 for the link!). After copying those files to the Talos boot partition, the PI booted correctly!

I'm unsure, but shouldn't this be a fix on Talos side? It appears to me that upstream blobs are working, while the ones shipped by Talos are outdated/causing issues with latest bootloader/firmware versions. I may be horribly wrong though.

samip5, 2025-03-27:
> I'm unsure, but shouldn't this be a fix on Talos side? It appears to me that upstream blobs are working, while the ones shipped by Talos are outdated/causing issues with latest bootloader/firmware versions. I may be horribly wrong though.

There hasn't been a new overlay release in a while, so even if the code in this repo would pull the right blobs, it's yet to find it's way to an Talos release.

coredump, 2025-03-27:
Just to chime in here, Raspberry Pi 4B, same problem/error as @AlessandroZanatta, did the same steps (updated the raspi using `raspi-update` and copied the files from the firmware to the talos boot partition) and that made talos boot at least

samip5, 2025-03-28:
> just didn't go to the monitor screen or anything so who knows

That's by design on SBCs

coredump, 2025-03-28:
Ok so after many tries, I wasn't able to get a Rpi 4 working. Those are some broad strokes from what I remember from the last few tries:

- Boot does't work with the standard 1.9.5 for metal-arm64
- After following the instructions [above ](https://github.com/siderolabs/sbc-raspberrypi/issues/43#issuecomment-2757054477)to download the rpi firmware and add it to the boot partition, it boots and I am able to execute `talosctl` commands against the IP address
- `get disks` only lists loopback devices of small size, `loop0`, `loop1` and etc
- Installing using any of those loops reboots to a screen where it says `firmware not found`, and if I turn the power off and on again, it goes back to maintenance mode
- Installing it on a external USB device works as it shows `/dev/sda` on the disk list, but same problem happens `firmware not found` even if I set the Pi to boot from USB

Some of those may be related to my inexperience with Talos itself, but I tried what I could think for debugging/troubleshooting and wasn't successful. Hopefully someone has better ideas or a fix appears in the next few versions.