siderolabs/sbc-raspberrypi #48: RPi4 Are Unable to Boot

chr0n1x, 2025-04-10:
I made a comment here, but I'm unsure if it's relevant so figured I should make an issue here.

https://github.com/siderolabs/talos/issues/9981#issuecomment-2791315599

Image Factory config:

```yaml
overlay:
    image: siderolabs/sbc-raspberrypi
    name: rpi_generic
customization:
    systemExtensions:
        officialExtensions:
            - siderolabs/iscsi-tools
            - siderolabs/util-linux-tools
```

my exact steps

1. use rpi imager 1.8.5 to flash the eeprom onto the card
2. insert card into rpi 4, confirm green LED is infinitely flashing, HDMI screen is completely green
3. download the image from the factory, `xz -d metal-arm64.raw.xz`
4. take SD card, `dd of=/dev/{sdX} if=metal-amd64.raw conv=fsync bs=4M`
5. validate that multiple partitions written via `lsblk`, searing for my `/dev/{sdX}`
6. insert SD card into rpi4. fancy boot screen w/ a QR code + rpi logo pops up
7. immediately goes to rainbow [screen of death?] 😄 

any tips on how I can further debug?

samip5, 2025-04-10:
Which bootloader version? Should be visible when you take the SD card out, and just apply power to it.
Should be listed somewhere on top of the screen.

samip5, 2025-04-10:
Tested with bootloader 8ba17717 / 2023/01/11, and 1.9.5 version of Talos and it boots absolutely fine.

https://factory.talos.dev/?arch=arm64&board=rpi_generic&cmdline-set=true&extensions=-&platform=metal&target=sbc&version=1.9.5 => https://factory.talos.dev/image/ee21ef4a5ef808a9b7484cc0dda0f25075021691c8c09a276591eedb638ea1f9/v1.9.5/metal-arm64.raw.xz

samip5, 2025-04-10:
> what can I do to figure that out?

Even without the SD-card inserted?

chr0n1x, 2025-04-10:
@samip5 oh, I did no try _just_ powering on the rpi without an SD card. doing that leaves me at the bootloader screen and it's looping, looking for the SD card.

chr0n1x, 2025-04-10:
@samip5 they're saying that there's a fix in `rpi-update`. sorry, I'm a bit lost here, but do you know how that fix would make it here downstream?

samip5, 2025-04-10:
> @samip5 oh, I did not try _just_ powering on the rpi without an SD card. doing that leaves me at the bootloader screen and it's looping, looking for the SD card.

So did you look at the screen and tell us which bootloader version it is?

chr0n1x, 2025-04-11:
@samip5 sorry, I might be slow here. I don't see a boot loader version on the screen.

samip5, 2025-04-11:
> @samip5 sorry, I might be slow here. I don't see a boot loader version on the screen.
> 
> edit: here's a photo
> 
> [image]

You might need to press Esc on keyboard for it.

chr0n1x, 2025-04-11:
ah ok. thanks for the tip. I had to unplug the power in order for esc to work

bootloader: `75c1e570` 2025/02/11
update-ts: 1739883497

samip5, 2025-04-11:
Yes, so there should be a newer firmware than that available.
One way is to boot RaspiOS Bookworm, and then run the update or use the rpi-imager for the same.

chr0n1x, 2025-04-15:
hey @samip5 sorry for the late followup. so Im not sure what I'm doing wrong.

1. I use the rpi bootloader, install raspiOS onto the SD card
2. ssh into the new installation
3. `apt update && apt full-upgrade && rpi-update`, reboot
4. `apt upgrade rpi-eeprom`,  `raspi-config` update

The version showing up in `rpi-eeprom-update` is still the feb 2025 version.

also, assuming that I get the latest bootloader on there (march 2025 or something?) am I supposed to be syncing the raw image to `/dev/sdc2` or something?

sorry, I'm quite lost. if there's any doc that I can reference for this please let me know so I can stop pestering you

samip5, 2025-04-15:
Okay, I was able to replicate this with Pi4, and 2025/02/11 bootloader,  blinking green led and the rainbow splash screen.

chr0n1x, 2025-04-15:
@samip5 nice! glad we're on the same page 😄 

any next steps that I can take? either for myself or to help out?

samip5, 2025-04-15:
> @samip5 nice! glad we're on the same page 😄 
> 
> any next steps that I can take? either for myself or to help out?

I ordered an UART debug device so I can easily get UART out and see if it will tell me what's wrong.

If you happen to have USB TO SERIAL thing, I would encourage you to see what it says on UART via GPIO pins. Might need some config.txt changes for getting more out of it tho.

chr0n1x, 2025-04-16:
have a link to that? I might have one lying around, don't know what that is by name though 😟

samip5, 2025-04-16:
> have a link to that? I might have one lying around, don't know what that is by name though 😟

https://www.raspberrypi.com/products/debug-probe/

samip5, 2025-04-16:
Tested with an older Pi4, running bootloader e608a69d / 2024/04/15 and it boots Talos 1.9.5 absolutely fine with the same SD card so it's definitely bootloader.

samip5, 2025-04-16:
Tested out 1.10.0-beta.0 just because, and it boots fine. :)

chr0n1x, 2025-04-16:
@samip5 awesome! how do I go about doing that? I probably missed the man page that explained updating to a prerelease(?) build

samip5, 2025-04-17:
> @samip5 awesome! how do I go about doing that? I probably missed the man page that explained updating to a prerelease(?) build

You go to factory.talos.dev and down-lasten the 1.10 beta that's in the bottom of the choices.

chr0n1x, 2025-04-17:
@samip5 it works!

the mini hdmi doesn't boot with anything though, is there an extension for displays that I'm unaware of in the 1.10.x series?

either way, thank you so much for your help and patience 🥰 I'll close.

closed 2025-04-17 by chr0n1x

samip5, 2025-04-19:
> @samip5 it works!
> 
> the mini hdmi doesn't boot with anything though, is there an extension for displays that I'm unaware of in the 1.10.x series?
> 
> either way, thank you so much for your help and patience 🥰 I'll close.

Mine did show things in mini-hdmi but closest to the power plug.